miwagner/ESP32-Arduino-CAN #8: can filter example does not work

ganeshkumarsw, 2019-04-25:


miwagner, 2019-05-07:
Hi,

have you respected the description?

  // Set CAN Filter
  // See in the SJA1000 Datasheet chapter "6.4.15 Acceptance filter"
  // and the APPLICATION NOTE AN97076 chapter "4.1.2 Acceptance Filter" 
  // for PeliCAN Mode

For test you can set the filter to:
  p_filter.ACR0 = 0;
  p_filter.ACR1 = 0;
  p_filter.ACR2 = 0;
  p_filter.ACR3 = 0;

  p_filter.AMR0 = 0xFF;
  p_filter.AMR1 = 0xFF;
  p_filter.AMR2 = 0xFF;
  p_filter.AMR3 = 0xFF;

and you should receive all messages.

Bye

Michael

t-c-o-d-e, 2019-10-25:
Can anyone please explain filtering. Or just give me an example how to set ACRs an d AMRs for eg. ID 360 and mask for first two bytes of packet.

Thanks in advance!

ganeshkumarsw, 2019-10-25:
```
void CAN_ConfigFilterterMask(uint32_t acceptance_code, bool extId)
{
   #define byte(x, y) ((uint8_t)(x >> (y * 8)))
    uint32_t acceptance_mask;
    CAN_filter_t filter;

    ESP32Can.CANStop();

    if (acceptance_code == 0xFFFFFFFF)
    {
        // No filter
        acceptance_mask = 0xFFFFFFFF;
        acceptance_code = 0;
    }
    else
    {
        if (extId == true)
        {
            acceptance_code = (acceptance_code << (32 - 29));
            acceptance_mask = ~(CAN_EXTD_ID_MASK << (32 - 29));
        }
        else
        {
            acceptance_code = (acceptance_code << (32 - 11));
            acceptance_mask = ~(CAN_STD_ID_MASK << (32 - 11));
        }
    }

    filter = {
        .FM = Single_Mode,
        .ACR0 = byte(acceptance_code, 3),
        .ACR1 = byte(acceptance_code, 2),
        .ACR2 = byte(acceptance_code, 1),
        .ACR3 = byte(acceptance_code, 0),
        .AMR0 = byte(acceptance_mask, 3),
        .AMR1 = byte(acceptance_mask, 2),
        .AMR2 = byte(acceptance_mask, 1),
        .AMR3 = byte(acceptance_mask, 0),
    };

    ESP32Can.CANConfigFilter(&filter);

    // Init CAN Module
    ESP32Can.CANInit();
}
```

t-c-o-d-e, 2019-10-25:
I can get all packets with:
p_filter.ACR0 = 0;
p_filter.ACR1 = 0;
p_filter.ACR2 = 0;
p_filter.ACR3 = 0;

p_filter.AMR0 = 0xFF;
p_filter.AMR1 = 0xFF;
p_filter.AMR2 = 0xFF;
p_filter.AMR3 = 0xFF;

But still can't filter specific ID. Can you please put ACR and AMR values for accepting only frames with  ID 360. I'm so confused :/.

THANK YOU!!!

t-c-o-d-e, 2019-10-25:
p_filter.ACR0 = 0x360;
p_filter.ACR1 = 0;
p_filter.ACR2 = 0;
p_filter.ACR3 = 0;

p_filter.AMR0 = 0x1F;
p_filter.AMR1 = 0xFF;
p_filter.AMR2 = 0xFF;
p_filter.AMR3 = 0xFF;

Gives me all frames with ID 36x.

kendo55, 2020-02-26:
two applications, running perfectly!

`*  TEST 1:  set filter to  0x6F2 --> Single_Mode
*  ---------------------------------------------
*  0x6F2 = 0b 1101 1110 010  --> ACR0 = 0xDE / ACR1 = 0x40 / ACR2 = 0 / ACR3 = 0
*                            --> AMR0 = 0/ AMR1 = 0b0001 1111 = 0x1F / AMR2 = 0xFF / AMR3 = 0xFF
*                            
*  TEST 2:  set filter to  0x6F2 and 0x102 --> Dual_Mode
*  -----------------------------------------------------
*  0x6F2 = 0b 1101 1110 010  -->  0b 1101 1110 0100 0000 --> 0xDE 0x40
*                                    --ACR0--- --ACR1---
*  0x102 = 0b 0000 1100 110  -->  0b 0010 0000 0100 0000 --> 0x20 0x40 
*                                    --ACR2--- --ACR3---
*                                    
*                                    0000 0000 0001 1111 0000 0000 0001 1111 --> 0x00 0x1F 0x00 0x1F
*                                    --AMR0--- --AMR1--- --AMR2--- --AMR3---
`

closed 2020-08-16 by miwagner