miwagner/ESP32-Arduino-CAN #10: ESP32can_basic example hangs after disconnect of TX-line

TLS1000, 2019-07-03:
Hi, I just tried the esp32_can basic example and it worked right-away; thanks!

BUT: when I disconnect the tx line temporarily communication doesn't resume, even after resetting via the 'EN' button on the doitV1 board doesn help. I can only get it back running after a new power-up 

I connected two ESP32 doitV1 boards together via the gpio2(TX) and gpio4(rx) and loaded the example sketch in each one

TLS1000, 2019-07-03:
update: it also hangs after disconnection of one the the two the CAN lines, which is real-life event.

allex1978, 2019-10-17:
Yes. it hangs after third send attempt to disconnected line. I looked line via logic analazer and it endlessly repeats sending the last packet to the line with a very short intervals.

jerryyhl, 2019-12-06:
Hi, I'm encountering this same issue, did anyone figure out a fix for this?

rocrail, 2020-07-03:
I solved this by replacing portMAX_DELAY with 500 in CAN.c
This will return after max. 500ms with zero if it could not be send.
`int CAN_write_frame(const CAN_frame_t *p_frame) {
	if (sem_tx_complete == NULL) {
		return -1;
	}
 
	// Write the frame to the controller
	CAN_write_frame_phy(p_frame);
 
	// wait for the frame tx to complete
	int rc = xSemaphoreTake(sem_tx_complete, 500);
	return rc;
}`

Satnet, 2021-04-18:
Doing this is a good solution and the proper way I would say

`xSemaphoreTake(sem_tx_complete, 10 * portTICK_PERIOD_MS);`

Itt uses tick in the low end code and adds 10 ms. So it's real time 10ms.
Sending a regular frame takes about 6 - 7ms so you could make it less if you want to. Works fine!
You'll find this inside the CAN.c file inside the function with name `CAN_write_frame(const CAN_frame_t *p_frame)`

katealhola, 2021-08-16:
Best solution would be having  int CAN_write_frame(const CAN_frame_t *p_frame, const tTickType_t xTicksToWait)

Current implementation where CAN_write_frame can hang infinite time is about blocker issue in real time system. Even having separate task just for sending and then having task hang infinite time is blocker issue. I have not find any other fix than patching this library because sem_tx_complete is static and there is no way from external code to check does function hang or no.

reshma26994, 2021-12-22:
using same library rx starts after that tx stops till restart and tx stops after rx starts