miwagner/ESP32-Arduino-CAN #11: Can bus hang

itsieber, 2019-08-14:
Hey Guys

I have a "Server" with 5 "Clients".

Everything works fine, but after a half Day, the Clients don't receive no packets any more.
In Serial i see the Server send this.

A Reset all clients not help.

A Reset only from the server would help.
A Softreset ESP.restart(); help not, only a physical reset.

Any one have a idea?

bharadwajinqore, 2020-06-02:
I am also having the same issue, I am using MCP2542 Click board, after sending the message controller hangs.
But my MCP2542 click board working with the PIC controller that i have already tested.

any suggestion greatly appreciated, .
Thanks in advance.

civic2016, 2020-12-22:
for softresets if you put .end() call before CAN init, that works for me.