miwagner/ESP32-Arduino-CAN #17: Cannot read CAN after writing CAN

ssummers-lime, 2020-02-04:
I try to run the basic example and it will read CAN data but as soon as the write function is called, it does not read CAN anymore. Also, I do not see the message being written out in my CAN viewer.

larktangle, 2020-02-15:
I have the same problem. Just using a generic ESP-WROOM-32 and SN65HVD230.

larktangle, 2020-02-21:
I have also tried swapping the SN65HVD230 for SN65HVD232, but still not working.

miwagner, 2020-02-25:
Hi,
is your construction as follows:

[image]

You need two bus participants and two 120Ohm resistor. Only with two boards the send and receive works.

Best regards

ssummers-lime, 2020-02-25:
Yes. My setup is correct. I am using the pcan usb device to simulate can data and read can data from the esp32. I never see any of the data that is supposed to be written out. I will also send one packet of data from the pcan device and the esp32 will receive it but it keeps reading the same data from the receive queue even though I only sent one message.

kendo55, 2020-02-26:
i have the same problem: 
please see in your CAN.c:
i think the problem is,
that CAN_write_frame() ist waiting after xSemaphoreTake(sem_tx_complete, portMAX_DELAY);
on a xSemaphoreGive, for ever!!!
In the CAN_isr() 
is the xSemaphoreGiveFromISR(),
but this will arrive, only if there is a error interrupt,
that will calling this CAN_isr() !
where you have the xSemaphoreGive()
in a "non_error_case"?

kendo55, 2020-02-26:
`a=B`

fffffffffff

`c=d`

kendo55, 2020-03-02:
addition to my last post:
 i have modified the CAN.c, to debug the  CAN_isr() with 
ets_printf("\n.....CAN-interrupt = 0x%02X\n", interrupt); 
And the Problem realy is, what 
after ESP32Can.CANWriteFrame(&tx_frame);
the following Int. appears:

1. CAN-interrupt = 0x80   -->BUS_ERR

back to main-loop,
 ESP32Can.CANWriteFrame(&tx_frame);
 the 2nd time 

2. CAN-interrupt = 0x04     --->CAN_IRQ_ERR  
immediatly followed
3. CAN-interrupt = 0x20   --->CAN_IRQ_ERR_PASSIVE

back to main-loop,
 ESP32Can.CANWriteFrame(&tx_frame);
 the 3rd time 

from now on it "hangs" :
--> becouse where is no CAN-Int. and no xSemaphoreGiveFromISR()

The solution will be, to check the "BUS_STATE" befor CANWriteFrame()

we would have to implement an error routine anyway.

kendo55, 2020-03-04:
The Problem was a bad series chips SN65HDV230 from china.
Sending was the problem. Receiving was o.k. !
I'm using now the MCP2562 driver!
Anyway, we should implement an error routine!

ssummers-lime, 2020-03-10:
Thanks for the help @kendo55. I just got the MCP2562 and all is working now.

closed 2020-03-10 by ssummers-lime