miwagner/ESP32-Arduino-CAN #19: DLC is always 8

romatetemadze, 2020-06-22:
Hello guys, I am trying this with Copperhilltech's ESP CAN module (https://copperhilltech.com/esp32-wifi-bluetooth-classic-ble-can-bus-module) and my issue is that every packet I receive is always of a size of 8 bytes which is not correct based on other tools I am sniffing the traffic with.

CAR - BMW 335 2007 (MSD80)
I am sending out: ID: 0x6F1 Data: 0x12, 0x02, 0x3E, 0x01
Where second byte (according to the protocol in use) is the size (2 bytes).

I am receiving this (correct response from DME):
"Got Frame from 0x612, DLC: 8, Data: 0xF1 0x01 0x7E 0xFF 0xFF 0xFF 0xFF 0xFF"

But I know for sure that second byte in a response is a message size (1 byte), and all of the FF's are somewhat extra. Is this how it works in general? Other libs (on atmgea328 for instance) would repost just 0xF1 0x01 0x7E as a payload.

Here is my code:

`#include <Arduino.h>
#include <ESP32CAN.h>
#include <CAN_config.h>

CAN_device_t CAN_cfg;               // CAN Config
unsigned long previousMillis = 0;   // will store last time a CAN Message was send
const int interval = 1000;          // interval at which send CAN Messages (milliseconds)
const int rx_queue_size = 10;       // Receive Queue size

void setup() {
  Serial.begin(115200);
  Serial.println("MHD PRO S1");
  CAN_cfg.speed = CAN_SPEED_500KBPS;
  CAN_cfg.tx_pin_id = GPIO_NUM_25;
  CAN_cfg.rx_pin_id = GPIO_NUM_26;
  CAN_cfg.rx_queue = xQueueCreate(rx_queue_size, sizeof(CAN_frame_t));
  // Init CAN Module
  ESP32Can.CANInit();
//  if (ESP32Can) {
//    Serial.println("Unable to initi CAN");
//  }
}

void loop() {

  CAN_frame_t rx_frame;

  unsigned long currentMillis = millis();

  // Receive next CAN frame from queue
  if (xQueueReceive(CAN_cfg.rx_queue, &rx_frame, 3 * portTICK_PERIOD_MS) == pdTRUE) {

    if (rx_frame.FIR.B.FF == CAN_frame_std) {
      printf("New standard frame");
    }
    else {
      printf("CAN::Received");
    }

    if (rx_frame.FIR.B.RTR == CAN_RTR) {
      printf(" RTR from 0x%08X, DLC %d\r\n", rx_frame.MsgID,  rx_frame.FIR.B.DLC);
    }
    else {
      printf(" from 0x%03X, DLC %d, Data ", rx_frame.MsgID,  rx_frame.FIR.B.DLC);
      for (int i = 0; i < rx_frame.FIR.B.DLC; i++) {
        printf("0x%02X ", rx_frame.data.u8[i]);
      }
      printf("\n");
    }
  }
  // Send CAN Message
  if (currentMillis - previousMillis >= interval) {
    //Serial.println("Send new msg");
    previousMillis = currentMillis;
    CAN_frame_t tx_frame;
    tx_frame.FIR.B.FF = CAN_frame_std;
    tx_frame.MsgID = 0x6F1;
    tx_frame.FIR.B.DLC = 4;
    tx_frame.data.u8[0] = 0x12;
    tx_frame.data.u8[1] = 0x02;
    tx_frame.data.u8[2] = 0x3E;
    tx_frame.data.u8[3] = 0x01;
    ESP32Can.CANWriteFrame(&tx_frame);
  }
}`

maxdd, 2021-05-04:
This is happening to me as well.
Could it be that the CAN DLC is really always 8?

hamza765, 2022-03-11:
Late response, i don't know if it's like this for all ECUs, but i've been working on the MED17 and there's always 0x55 padding added to each frame to ensure a _frame_ length of 8. Note, that means that i'll have 8 bytes of data in the array, but not that i have 8 bytes to _usable_ data. My DLC dictates how much of the frame is actual data. 

For example, say i have a frame: `02 3E 00 55 55 55 55 55`
Take the first byte `02`
The `0` indicates it's a single frame
The `2` indicates that there are 2 bytes of usable data
`3E 00` are those important bytes of data

Hope this helps