miwagner/ESP32-Arduino-CAN #22: No ACK mode Needed

retsel89, 2020-08-13:
Hello, I tried the library with a circuit with a transceiver and it is working perfectly, but I have a non-invasive circuit, without a transceiver that works without ACK so I need to initialize the CAN Bus without ACK, the CANInit function initializes it in normal mode, I have been trying some variants but I can't, I think the key would be in this line:
// set to normal mode
MODULE_CAN-> OCR.B.OCMODE = __CAN_OC_NOM;

and with this structure:

unsigned int RM: 1; / ** <\ brief MOD.0 Reset Mode * /
unsigned int LOM: 1; / ** <\ brief MOD.1 Listen Only Mode * /
unsigned int STM: 1; / ** <\ brief MOD.2 Self Test Mode * /
unsigned int AFM: 1; / ** <\ brief MOD.3 Acceptance Filter Mode * /
unsigned int SM: 1; / ** <\ brief MOD.4 Sleep Mode * /
unsigned int reserved_27: 27; / ** <\ brief \ internal Reserved * /

Could you give me some help?

* Note: I made a measurement on the Tx and Rx pins of the esp32 and it has a perfect square signal, apparently the Hardware is OK, but the main requirement is to use the CAN bus without ACK

miwagner, 2020-08-16:
Hi,

I read in the data sheet that the "Listen only" mode list should be activated in this way:

In the file CAN.c replace in line 260 the code "MODULE_CAN->MOD.B.RM = 0;" by "MODULE_CAN->MOD.B.LOM = 1;".

That should put the controller in Listen Only Mode and leave Reset Mode. Please try out.

bye

retsel89, 2020-08-18:
Hello,
I replaced line 260 "MODULE_CAN->MOD.B.RM = 0;" by "MODULE_CAN->MOD.B.LOM = 1;" and it didn't work, so I decided with the Arduino framework to use the native IDF libraries to initialize the can in read-only mode and it worked perfect, I leave the code that I use which is basically the IDF read-only example.

need include
#include "driver/can.h"

can_general_config_t g_config;

void begin_can(void){
    g_config.mode = CAN_MODE_LISTEN_ONLY;
    g_config.tx_io = GPIO_NUM_21;
    g_config.rx_io = GPIO_NUM_22;
    g_config.tx_queue_len = 0;
    g_config.rx_queue_len = 256;
    g_config.alerts_enabled = CAN_ALERT_NONE;
    g_config.clkout_divider = 0;

    can_queue = xQueueCreate(can_queu_size,sizeof(Can_frame_mod));
    //Install and start CAN driver
    Serial.println("begin CAN");
    can_driver_install(&g_config, &t_config, &f_config);
    can_start();
    Serial.println("CAN started");

    xTaskCreatePinnedToCore(can_rx_task, "CAN_rx", 4096, NULL, 9, NULL, tskNO_AFFINITY);
}