miwagner/ESP32-Arduino-CAN #23: Update for ESP32-WROOM32E - V3

Juantxor, 2020-10-29:
ESP32 ECO3 added a configurable bit that will cause CAN bus speeds to be halved.
https://www.esp32.com/viewtopic.php?t=15581
https://github.com/espressif/esp-idf/commit/03d5742e110d2d5a8fbdf60ad9fcf894d3f98eb5

fleetsu, 2020-10-31:
Confirmed,  I just had the same issue, when trying to use a 32E module to talk 500kbps Network. I had to set the speed to 1000kbps before it would work.

ScheintodX, 2021-04-28:
Same here for an Olimex esp32-evb Rev-I board.

field172, 2021-05-27:
Very interesting! Thank you for sharing this. I'm experiencing similar issue, with with WROOM but similar, mine is TTGO ESP32 240MHz WiFi/BT 16MB flash.  
The CAN I'm communicating with is confirmed 500Kbps (other devices successfully tx & rx at 500Kbps, and documentation confirm 500Kbps), however, for any working communications with ESP32CAN I must set CAN to 1000Kbps, and even so an able to only receive messages (so far no success transmitting).  
Same issue trying to connect with known 125Kbps CAN, my ESP32CAN only working to receive when set to 250Kbps, no success with tx.

field172, 2021-05-27:
I can confirm that my esp32 TTGO is now receiving at correct speed set to 500Kbps, fixed by changing CAN.c library file, 0xff to 0xef:
	// enable all interrupts
	MODULE_CAN->IER.U = 0xef;