miwagner/ESP32-Arduino-CAN #27: ESP32Can.CANInit hang my application

pixsysedp, 2021-03-30:
Hi, I am using this library on an ESP32 project where other peripherals are used.
In particular I use the ADC and the two UARTs with modbus protocol.
If in "setup()" I call the "ESP32Can.CANInit ();" function but there is no CAN master, the ESP32 is blocked, for example, no uart response are possible.
Is there any workaround? I have no way of knowing in advance when the master will be connected.

pixsysedp, 2021-03-31:
I placed a "LED toggle" in the main loop and it stops if I turn off my can master.
Now I did a little test by calling CANStop () on the slave before turning off my master but I didn't see any difference. Is there a "CANDeInit ()"?

Satnet, 2021-04-18:
It is blocked because of this and not the init function:
`xSemaphoreTake(sem_tx_complete, portMAX_DELAY);` inside CAN.c within the function `CAN_write_frame(const CAN_frame_t *p_frame)` this waits for the CAN to send and because it can't it hangs due to MAX_DELAY as seen above. Change this to:

`xSemaphoreTake(sem_tx_complete, 10 * portTICK_PERIOD_MS);` where you can lower 10 to 6 or 7 as that what it takes to send a frame. I have it at 10 which is 10ms, works like a charm. If it can't send, then after 10ms it'll move on.

pswired, 2022-03-13:
I encountered this problem, and found that setting a 10ms timeout worked at higher speed (125kbps and above) baud rates, but wasn't enough time for lower baud rates. A more elegant solution might be to enable SJA1000 self test / loopback mode so that transmissions are automatically self-acknowledged. You can insert this into CAN_init() of CAN.c:

```
	// set to self-test/loopback mode so that an open bus is acceptable
	MODULE_CAN->MOD.B.STM = 1;
```