miwagner/ESP32-Arduino-CAN #35: Slow speed CAN BUS 20KBPS

andreadalessandroaptar, 2022-03-07:
Hello guys,

I'm trying to use a custom board
https://www.tindie.com/products/fusion/can32-an-esp32-dev-board-with-can-bus-v21/

It has an official ESP32-Wroom-32UE and a MAX3051 CAN transceiver.

I would like to use it with my Daikin heater, unfortunately this can bus is very slow 20KBPS, and the board doesn't receive anything. 
I've also tried to modify the the speed in the code cpp files in order to be able to use this speed with no luck, because I've read there is something related to multipliers which is not gonna work.

Just to be sure about the issue, I've tested successfully with the help of another ESP32 + MCP2515 board (of course slowed down to 20KBPS).
There's no way to raise a little bit this rate.

Is anyone here which got same behaviour ? There's really no option ?

Thanks!

pswired, 2022-03-13:
This is a limitation of the hardware in <v2 ESP32 devices. See here for details:

https://esp32.com/viewtopic.php?t=2142

MaKo-RS, 2022-03-18:
The slowest possible baudrate in ESP32 revision lower than V2 is 25kBit/s. This is due to the APB-clock and the 6-Bit prescaler.
With ESP32 rev2 or now also available rev3 it should be possible. But i didn't find any lib for arduino supporting the new feature. The API reference of ESP-IDF (v4.1 2020-03-10):
_"The Baudrate Prescaler is used to determine the period of each time quanta by dividing the CAN controller’s source clock (80 MHz APB clock). The brp can be any even number from 2 to 128. If the ESP32 is a revision 2 or later chip, the brp will also support any multiple of 4 from 132 to 256, and can be enabled by setting the [CONFIG_ESP32_REV_MIN] to revision 2 or higher."_

Does anyone know how to enable it with the arduino lib?
Thanks!