miwagner/ESP32-Arduino-CAN #47: This library is outdated!

morcibacsi, 2022-10-10:
**You shouldn't use this library** as it was written in the early days when the ESP-IDF was different. Since then the Arduino core for ESP32 has a very good AND reliable interface. Even on a 125 kbps bus this library isn't reliable and you are going on a wild goose chase with mysterious errors on your CAN bus.

You need only a few lines of code to have CAN working. Please read the official ESP32 documentation here: [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html#examples](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html#examples)

If that is TL;DR, then here is the gist of it: 

```cpp
SemaphoreHandle_t canSemaphore;

void InitCAN()
{
    canSemaphore = xSemaphoreCreateMutex();
    can_general_config_t g_config = {.mode = CAN_MODE_NORMAL,
                                     .tx_io = (gpio_num_t)txPin, .rx_io = (gpio_num_t)rxPin,
                                     .clkout_io = CAN_IO_UNUSED, .bus_off_io = CAN_IO_UNUSED,
                                     .tx_queue_len = 10, .rx_queue_len = 10,
                                     .alerts_enabled = CAN_ALERT_NONE,  .clkout_divider = 0,
                                     .intr_flags = ESP_INTR_FLAG_LEVEL1};

    can_timing_config_t t_config = CAN_TIMING_CONFIG_125KBITS();
    can_filter_config_t f_config = CAN_FILTER_CONFIG_ACCEPT_ALL();

    esp_err_t result = can_driver_install(&g_config, &t_config, &f_config);
}

uint8_t SendMessage(uint16_t canId, uint8_t ext, uint8_t sizeOfByteArray, uint8_t *byteArray)
{
    can_message_t message;
    message.identifier = canId;
    message.flags = CAN_MSG_FLAG_NONE;
    message.data_length_code = sizeOfByteArray;
    for (int i = 0; i < sizeOfByteArray; i++) {
        message.data[i] = byteArray[i];
    }

    uint8_t result = 0;

    if (xSemaphoreTake(canSemaphore, portMAX_DELAY) == pdTRUE)
    {
        if (can_transmit(&message, pdMS_TO_TICKS(10)) == ESP_OK) {
            result = 1;
        } else {
            result = 0;
        }
        xSemaphoreGive(canSemaphore);
    }
    return result;
}

void ReadMessage(uint16_t *canId, uint8_t *len, uint8_t *buf)
{
    can_message_t message;
    if (can_receive(&message, pdMS_TO_TICKS(10)) == ESP_OK) {
        if (message.flags == CAN_MSG_FLAG_NONE || message.flags == CAN_MSG_FLAG_SS)
        {
            *canId = message.identifier;
            *len = message.data_length_code;
            for (int i = 0; i < message.data_length_code; i++)
            {
                buf[i] = message.data[i];
            }
        }
    } else {
        return;
    }
}
```

latonita, 2022-10-19:
Nice note.  Just a quick question - Is there a calculator on how to set up can speeds?
For example 83.3kbps is often used in automotive industry. I dont see predefined option in that esp32 header

morcibacsi, 2022-10-19:
I found some code on the esp32 forum which should calculate the correct timing for whatever speed, I haven't tested it though: https://www.esp32.com/viewtopic.php?t=23719

MeisterQ, 2022-10-27:
@morcibacsi 
This is not working.

Ive tried everything. Can you give me a "readymade" code please?

morcibacsi, 2022-10-27:
@MeisterQ 
What is not working? It doesn't compile, or nothing is sent on the bus? I just updated the code above with one line in the InitCAN method, maybe that was what you were missing: 

```cpp
result = can_start();
```

I have a library with this code in one of my project which you can find here: https://github.com/morcibacsi/PSAWifiDisplayControl/blob/master/PSAWifiDisplayControl/PSAWifiDisplayControlMain.cpp#L122

There you can see how it should be initialized and how you can write to the CAN bus.

But I **_strongly_** recommend to read the official documentation which I linked above, as it has everything which is necessary, and it covers more than my library as I only implemented which was necessary for my use case.

lolimpol, 2022-11-01:
So it just works like this in the Arduino IDE?
what about the ESP32-S2? Not that I particularly need it, but newer is always more fun right?

morcibacsi, 2022-11-02:
> So it just works like this in the Arduino IDE?

Yes it works like this. 

> what about the ESP32-S2?

I don't have such a device, but why don't you try it? It can be checked in a few minutes. Also you can always check the official documentation for the S2 here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/twai.html#examples It is like 70 lines altogether, and the documentation is very good. Takes much less time to understand it than struggling to find a "ready-made" solution on google. I haven't checked it line-by-line, but I bet it is the same code as for the ESP32 (it doesn't make sense to be different as it would make it harder to maintain their codebase).

lolimpol, 2022-11-02:
That would make sense if I had one haha.
I'm trying to design a board for my car with it though, so I need to find something that just works. And my knowledge of Arduino is limited as hell. So I think I'll stick with the ESP32, as the chance that I'll get it working is higher (I think)

japui-coder, 2022-11-03:
This library is outdated! #47